osqzss/gps-sdr-sim #378: CAN this tools Spoof  newer gps receivers?

nxydyxsj123456, 2023-07-03:
I use hackrf, I use TCXO, I can successfully spoof old android phones, but I can't spoof 2019 samsung s10+, is it a setting problem or is this tool not able to spoof newer gps receivers

Imtiaz08, 2023-07-05:
It looks like the latest smartphone devices aren't solely relying on GNSS receiver for the calculation of position. There are multiple aiding sensors such as network location, built-in IMU, even Google smartphones such as Pixel use GLP services which creates a cache of the user position based on the processed information obtained from the devices within its vicinity. 

All these technologies make the smartphone somewhat resilient to basic type of spoofing attacks. Have you tried turning off the auto update time and date off and restarted the device before transmitting the signal? When you do this the device gets into cold start mode meaning it is more vulnerable to attacks now. 

However, the chances are still pretty low, can you please share the data of your old smartphones during the attack in an open sky environment? I am working on something but due to lack of hardware and data I am not able to do it.

psmitty7373, 2023-07-05:
I've had success with more modern phones.  Specifically clearing A-GPS data and blackholing the associated A-GPS domains helps great.  Additionally, you have to make sure the clock matches what your GPS data is sending.

For Pixel devices, I blackhole:
xtrapath1.izatcloud.net
xtrapath2.izatcloud.net
xtrapath3.izatcloud.net

Imtiaz08, 2023-07-05:
> I've had success with more modern phones.  Specifically clearing A-GPS data and blackholing the associated A-GPS domains helps greatly.  Additionally, you have to make sure the clock matches what your GPS data is sending.
> 
> For Pixel devices, I blackhole:
> xtrapath1.izatcloud.net
> xtrapath2.izatcloud.net
> xtrapath3.izatcloud.net

Can you please explain what transmitting front end are you using? I have a hackRF one without TCXO, also what about black holing the associated data? Never heard about that before...

psmitty7373, 2023-07-05:
Using a hackrf with txco. You should really invest in the crystal. It's
very inexpensive and definitely helped.

I'm using DNS black holing of the above domains so the phone cannot
redownload assisted gps data. If it has the agps data it is very difficult
to get a gps lock.  You can use one of many gps apps to clear the agps
cache or reboot the phone.

nxydyxsj123456, 2023-07-10:
I can success cheat old phone ，but in my samsung S10+,  10 satellite in view but 0 in use ,I have tried  txco with 0.1~0.5ppm, open airplane mode,clear AGPS,and reboot my phone.

psmitty7373, 2023-07-10:
Does the phone clock match the spoofed GPS time and date used in your apoapsis data?  If it does, then the S10+ might have some additional spoof detection.

nxydyxsj123456, 2023-07-10:
> It looks like the latest smartphone devices aren't solely relying on GNSS receiver for the calculation of position. There are multiple aiding sensors such as network location, built-in IMU, even [...]

i  dont konw what the data mean   i use old chinese  xiaomi phone and success

XT000601, 2024-12-19:
> 我尝试了过去时间和实时攻击，但在三星和 GPS 天线上均未成功，廉价的 hackrf 可以完成这项工作吗？我有更昂贵的设备，例如 x310 n210 b210，但没有找到它们的代码。我想知道这是反作弊问题还是我的设备问题。

well,my b210 is the same,don't conut on it.