osqzss/gps-sdr-sim #381: It suddenly can't work in the afternoon, why?

863356170, 2023-08-02:
This morning, I used gps-sdr-sim with my bladeRF xA9 and successfully found my phone and drone in the fake location. However, when I tried to do the same thing just 4 hours later, I failed. I think I changed nothing, but now my phone could find those satellites and even tell me it has used them while no positioning.
I used today's brdc file and have tried to use default time settings and the real time, whose results were the same. Neither static nor dynamic mode worked, too (while they both worked well in the morning). My phone is 7 years old (forgot its model) and worked well last time I used gps-sdr-sim. My drone is DJI mini 2 SE, could work this morning (even though I couldn't find it on the map, I knew it came to no fly zone). And I used GPS Test to check my achievement.
I couldn't find out the reason, please help. I am not a native English speaker so forgive my bad English.

863356170, 2023-08-03:
Today I tried again and ... it sometimes worked well but most of time can't work. I just kept receiving message, use them but no fix. Still want to know why. 
I'd like to know may it because I left my BladeRF connected to computer when having lunch(about two hours) and it was damaged? How can I check it?

jvictors-tp, 2023-10-05:
It does not make sense to me that this tool would not work at different times of the day. It should work all the time, or none of the time. I suspect that there is another explanation here. I don't think that there is any correlation between this tool and the afternoon.

I'm not getting enough information from your description to have any other ideas. From your second comment, it seems like your phone is not getting a GPS lock using this tool, and your BladeRF might be damaged you said. I know that you opened this issue several months ago, so I would like to ask: did you get this tool working for you? Did your phone acquire a lock? Was your BladeRF damaged?

(I hope my response makes sense to you, I tried to draw out my thoughts and sentences to help with the language barrier)

863356170, 2023-10-30:
Sorry for the late reply, I had to do other work after asking the question and did not continue to try to use this tool.
My description may not be clear, I'll try to explain it again and give my current situation.
The first time I used the tool, it worked well, successfully tricking my phone into thinking it was in another city, and it had the same effect in multiple tests. At that time, I used the latest ephemeris downloaded from CDDIS, used the -l option to enter the latitude and longitude, and the launch parameters completely referred to the bladerf parameters given in the instructions (but deleted set txvga1 -25 and the following two lines, they seemed not applicable in bladerf xA9).
I don’t think I made any changes after that, but I couldn't use the tool to change the location information of my phone again that afternoon (time has nothing to do with this, just narrative), which seems to mean that the initial success was unstable and just a fluke. .
After posting the question, I tried adding an attenuator fo about 20dB between the bladerf and the antenna, since other issues mentioned that GPS receivers are designed to receive very small signals, and excessive power may cause failure. However, this method is not suitable for my problem.
My colleague suggested that I add an amplifier between the bladerf and the antenna, so I connected an amplifier with a gain of about 20dB, adjusted the gain to the maximum, and got good results in subsequent tests: I changed the phone's position. I solved the problem for now, although the method is a little different from my understanding of a GPS receiver.
jvictors-tp, thanks for your reply, I think I've got this tool working for me and bladerf isn't broken, I'm just not using it correctly. You also made me realize that my ability to express myself is terrible. Thank you very much. It makes sense for me to realize this.

closed 2023-10-30 by 863356170

jvictors-tp, 2023-10-31:
No problem, thanks for the follow-up!