osqzss/gps-sdr-sim #382: Possible to create consistent SNR/CN0 simulation?

antmerlino, 2023-08-30:
Hello,

I am trying to use this software to test a GPS receiver in manufacturing.  To do this, I am trying to generate satellite signals with consistent and stable SNR values.  I have tried fixing the gain to 100 as mentioned in other posts, as well as disabling ionospheric delays.  When the simulation first starts, I get a very flat set of SNR values across all the satellites, but then as the simulation progresses, I start getting increasing disparities between the satellites. 

I am wondering if you could point me in the right direction as to what is causing this behavior.  I'd be happy to contribute any improvements back and also willing to make a donation for help supporting this. 

Thanks,
Anthony

osqzss, 2023-09-02:
I added the "-p" option to disable path loss and maintain a constant power level. It simply fixes the gain at 128 for all visible satellites. I tested the generated I/Q samples with a u-blox receiver, and there were no disparities between the satellites. If the signal level is too strong, it can lead to cross-correlations among the satellites. Please consider inserting appropriate attenuation in the RF line.

antmerlino, 2023-09-06:
Hi @osqzss. 

>  If the signal level is too strong, it can lead to cross-correlations among the satellites.

This was ultimately my issue.  I had 3x 20dB attenuators and a DC Block coming from the HackRF One, but that still wasn't enough.  Adding a 4th attenuator fixed the cross-correlation issue and gives me consistent SNRs.  

Thanks for the help!

closed 2023-09-06 by antmerlino