osqzss/gps-sdr-sim #387: lose fix every a few seconds on DJI mini 2 se

863356170, 2023-11-07:
I want to use a bladerf xA9 to spoof a DJI mini 2 se.
I have been able to make the drone "think" it is in the wrong place, but I encountered a strange phenomenon: the drone would lose its fix every tens of seconds, and no satellites could be seen on the DJI app at that time. Although the drone can fix again in a few seconds, I hope to completely avoid this interruption and make the fix more stable.
I once thought that this was related to the switching between frames of the GPS signal, but I searched and found that the length of the frame was 30 seconds, and a rough estimate showed that the interruption interval was more than 40 seconds, which seemed to rule out this possibility.
What could be the cause of what I'm experiencing? Can I avoid interruptions and keep the drone in a spoofed state without having to constantly re-lock? How to do?

863356170, 2023-11-07:
I find my title wrong with my translator ... I mean tens of seconds in fact. And I tested inside, I think there was no real GPS signal. I use gps-sdr-sim to create .bin file and use my bladerf to transmit it.

osqzss, 2023-11-11:
LEO BODNAR's Mini Precision GPS Reference Clock is a good and reasonable GPSDO.
https://www.leobodnar.com/shop/index.php?main_page=product_info&cPath=107&products_id=301&zenid=65c5ae1b8bc80efae25c25e14c84e996

mohmmadd, 2024-01-13:
> ok, I saw the issue of frequency offset caused by bladerf 2.0's clock mentioned in #361, maybe I'm experiencing the same problem. I don't have an external oscillator, so I will first try calibrating the clock with kalibrate...will update my progress next week.

any update about your progress?