osqzss/gps-sdr-sim #391: Help I cant get a fix

Dylan-Evins, 2023-12-22:
I'm attempting to spoof a BU-353S4 GPS Receiver using my Hack-RF One however I cant seem to get it to get a fix. From my very limited knowledge and what I've read on here could this be a clock issue or is there something else that can be done? Here are the commands im using from start to finish and the results of the test. Any help is appreciated, thank you. 

sudo git clone https://github.com/osqzss/gps-sdr-sim.git    
sudo gcc gpssim.c -lm -O3 -o gps-sdr-sim 
sudo ./gps-sdr-sim -b 8 -e brdc3170.23n -l 52.52000800000000,-13.40500000000000, 100 
hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0
[image]

osqzss, 2023-12-22:
Please do not leave a space between a comma and the height in the -l option. gps-sdr-sim misunderstood it was another argument. You can try to use the default location without the -l option and see how it goes.

Dylan-Evins, 2023-12-23:
> Please do not leave a space between a comma and the height in the -l option. gps-sdr-sim misunderstood it was another argument. You can try to use the default location without the -l option and see how it goes.

Thanks for the reply! I fixed the space and tried the default location and then also tried spoofing a location nearby the correct gps location and neither worked. Any other suggestions?