osqzss/gps-sdr-sim #392: Generate GLONASS signals

Gerald0217, 2024-01-09:
Dear expert,

The specific query I have is: How to generate GLONASS signals? I am interested in understanding the technical aspects and processes involved in generating signals for GLONASS navigation.

If you could spare some time to share your expertise on this matter, I would greatly appreciate it. Any resources, references, or guidance you can provide will be immensely helpful.

Thank you very much for considering my inquiry. I look forward to hearing from you.

Best regards

osqzss, 2024-03-26:
Everything you need to know is written in the ICD: https://www.unavco.org/help/glossary/docs/ICD_GLONASS_4.0_(1998)_en.pdf

danusha2345, 2024-11-01:
**Good afternoon. Make a program to simulate glonass, please.**