osqzss/gps-sdr-sim #393: Troubleshooting GPS Spoofing with HackRF on Android

Petros04, 2024-01-23:
Hi everyone. My name is Petros, and I am learning at NPUA (in Armenia).
I have been working on GPS spoofing using HackRF for over 6 months, specifically focusing on GPS spoofing with noise and without noise. I have successfully spoofed two GPS receivers (BE 880 GPS and RY836AI) using the latest version of gps-sdr-sim. However, when I tried to spoof my Samsung S22 phone, I encountered inconsistent results. Sometimes the spoofing worked, but other times it did not. I initially suspected the issue might be with hackrf_transfer and attempted to use GNU Radio to transfer gpssim.bin with various configurations. Unfortunately, I haven't found the correct configuration for GPS spoofing that consistently works on my Samsung S22 phone. I have already disabled Wi-Fi and Bluetooth scanning on the phone. Could you please help me troubleshoot this problem, or let me know what could be causing the spoofing to not work on my Samsung S22 phone?

bin genereating example:  [sudo ./gps-sdr-sim -b 8 -s 2600000 -e brdc0230.24n -l' 40.48721406299335,44.51516122914086,100 -T 2024/01/23,15:07:07 -d 400 -p]

herrameise, 2024-02-07:
You probably need to change the "Scale Factor" in the "IChar to Complex" block to 128. This will take the complex char IQ sample values (currently ranging from -128 to 127) and map them into the normal range for complex float IQ samples (-1.0 to 1.0). I can't say that will solve your problem with the Samsung phone though

herrameise, 2024-02-07:
also I'm not familiar with the oscomocom blocks, but the actual GPS L1 A/C signal bandwidth is something like 2 MHz (not 500 KHz as you have)

kings-way, 2024-06-28:
Same here, failed to spoof multiple android devices.
(both hackRF and USRP B210)

victor-vu-max, 2024-08-19:
yeah i have also been failure of transmiting spoofing GPS signal

victor-vu-max, 2024-08-19:
> Same here, failed to spoof multiple android devices. (both hackRF and USRP B210)

Same here

victor-vu-max, 2024-08-19:
[image]
[image]
the location on Androin devices can not change. I think AGPS can not be spoofing

cdhmanning, 2025-01-14:
Spoofing can be a challenge if the device has augmentation. If the device knows where it is, knows the time and knows what satellites it should be listening to then it might ignore the signal you are spoofing with.

Also, it is worth turning off other constellations to make sure the receiver is ONLY listening to GPS. If you're spoofing GPS and the receiver is getting correct position information via, say, Beidou, then maybe it will ignore your GPS signal.