osqzss/gps-sdr-sim #395: options

NAVIGATOR19, 2024-03-25:
I have a question about these options
-u <user_motion> User motion file in ECEF x, y, z format (dynamic mode)
-x <user_motion> User motion file in lat, lon, height format (dynamic mode)
-g <nmea_gga>    NMEA GGA stream (dynamic mode)
Can files be generated on the fly, or are they cyclically read from the beginning to the end, and nothing can be added to them during transmission?

osqzss, 2024-03-26:
Nothing can be added to the motion files during execution.

NAVIGATOR19, 2024-03-27:
> Nothing can be added to the motion files during execution.
How can I dynamically change the coordinates transmitted via HackRF?

NAVIGATOR19, 2024-03-27:
How can I dynamically change the coordinates transmitted via HackRF?