osqzss/gps-sdr-sim #404: USRP under flow issue.

yosefzh, 2024-07-16:
Hello. 
im using a USRPX310 on a windows device , i don't know if there is a easier way to do this on windows but i tried and failed.

so i downloaded a linux virtual machine and used it to work after solving multiple errors it runs the program, but it spams UUUUUUUUU
an under-flow issue i don't know how to solve it , as you can see here, then i tried the other way using Tx sample from example but it didn't work also for some reason when i looked to the file sample from example was empty so i don't know how should i run it 
[image]

Sselion, 2024-07-22:
Hi, i'm using a USRP B205 mini.
An Underflow 'U' occure when USRP has not enough samples to send. This can happen when when the thread responsible for sending samples to the device is suspended at a critical moment, causing a delay. Or  it can occur if you do not read the data from the file quickly enough. I don't know how familiar are you with generated data format. With basic setup SC16 is it 2 byte per I and Q sample and 2.6*10^6 samples per second. That mean 2 * 2 * 2.6*10^6 =  10.4 MB/s.

In my case I have to change priority of process to realtime and maximalise thread priority in my C# script.

Try learn more in USRP manual: https://files.ettus.com/manual/page_general.html

yosefzh, 2024-07-22:
Hello, Thank you for getting back to me ! 

i worked around that issue and spent alot of time compiling the app in windows and now it runs ! but the problem is im getting alot of satlites in view . 
but no one is in use . 
do you have any idea why ? im trying to figure that now

yosefzh, 2024-07-22:
as you can see here alot of satellites signals in view but 0 in use 
[
[image]
](url)

zhaoyongfa, 2024-09-10:
> 您好，谢谢您回复我！
> 
> 我解决了这个问题，花了很多时间在 Windows 中编译应用程序，现在它可以运行了！但问题是我看到了很多卫星。 但没有人在使用。 你知道为什么吗？我现在正试图弄清楚

能否分享下代码？

shuwang1, 2024-09-10:
Have you checked these satellites are valid, supposed to be visible, or not?

xixi88888888, 2024-09-10:
> 您好，谢谢您回复我！
> 
> 我解决了这个问题，花了很多时间在 Windows 中编译应用程序，现在它可以运行了！但问题是我看到了很多卫星。 但没有人在使用。 你知道为什么吗？我现在正试图弄清楚

Could you please share your solution or code? I would greatly appreciate it.

xixi88888888, 2024-09-10:
> 你好，我正在使用 USRP B205 mini。 当 USRP 没有足够的样本发送时，就会出现下溢“U”。当负责向设备发送样本的线程在关键时刻被暂停，从而导致延迟时，就会发生这种情况。或者，如果您没有足够快地从文件中读取数据，也会发生这种情况。我不知道您对生成的数据格式有多熟悉。基本设置 SC16 是每个 I 和 Q 样本 2 字节，_每秒 2.6 10^6 个样本。这意味着 2 * 2 * 2.6_ 10^6 = 10.4 MB/s。
> 
> 就我而言，我必须将进程的优先级更改为实时，并在我的 C# 脚本中最大化线程优先级。
> 
> 尝试在 USRP 手册中了解更多信息：https://files.ettus.com/manual/page_general.html

Could you please share your solution or code? I would greatly appreciate it.

XT000601, 2024-12-17:
> 正如你在这里看到的，有很多卫星信号，但实际使用的却为零 [ ![IMG_1690](https://private-user-images.githubusercontent.com/66050542/351001980-b69f09f7-6363-4908-936a-54f0425cbcf5.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ0NDA1MTYsIm5iZiI6MTczNDQ0MDIxNiwicGF0aCI6Ii82NjA1MDU0Mi8zNTEwMDE5ODAtYjY5ZjA5ZjctNjM2My00OTA4LTkzNmEtNTRmMDQyNWNiY2Y1LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxN1QxMjU2NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iY2ZhYWZmN2UzOWViMzg5YjE2Y2RhOWE1N2VkMjQ4MDAyZjQxYzUzOTNiM2ZjMmVkZGQ1YjgzZDlkZTRkNTUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.vjHigBiSmki16exML0GBNYNUXieceFXQ64XBpv3s3cs) ](url)

I have this problem too,have you solve it?