osqzss/gps-sdr-sim #405: BeiDou B1I signal

cehuidong, 2024-07-24:
Has any expert modeled the Beidou B1I signal. Can the BeiDou B1I signal be modulated similarly to GPS? I would appreciate if anyone can guide or provide references. Thanks!

le0nidus, 2024-08-22:
hey, someone started working on it in a similar project. looks like he stopped..
https://github.com/yangfan852219770/beidou-sdr-sim

zhaoyongfa, 2024-09-09:
可以一样调制，仅仅B1I可看做bpsk

cehuidong, 2024-09-09:
> 可以一样调制，仅仅B1I可看做bpsk

非常感谢。我根据项目：https://github.com/yangfan852219770/beidou-sdr-sim和GPS-SDR-SIM调制出了B1I信号，但是我在使用动态坐标生成的信号会出现导航电文读错的情况，但静态坐标不会，而且计算的伪距错误
[image]
。

cehuidong, 2024-09-09:
> 可以一样调制，仅仅B1I可看做bpsk

非常感谢。我根据项目： (https://github.com/yangfan852219770/beidou-sdr-sim)  和GPS-SDR-SIM调制出了B1I信号，但是我在使用动态坐标生成的信号会出现导航电文读错的情况，但静态坐标不会，而且计算的伪距错误。
[image]

xixi88888888, 2024-09-10:
> > 可以一样调制，仅仅B1I可看做bpsk
> 
> 非常感谢。我根据项目： (https://github.com/yangfan852219770/beidou-sdr-sim) 和GPS-SDR-SIM调制出了B1I信号，但是我在使用动态坐标生成的信号会出现导航电文读错的情况，但静态坐标不会，而且计算的伪距错误。 ![屏幕截图 2024-08-28 195347](https://private-user-images.githubusercontent.com/115219834/365685861-ac14f5fb-d111-4832-bf00-939b0ef9c875.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5MzE1NzEsIm5iZiI6MTcyNTkzMTI3MSwicGF0aCI6Ii8xMTUyMTk4MzQvMzY1Njg1ODYxLWFjMTRmNWZiLWQxMTEtNDgzMi1iZjAwLTkzOWIwZWY5Yzg3NS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQwMTIxMTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZTQ3YjQ1YjdlMmJiOTZlNzk4NDQzNmIzZjgxMGYyZjdmNmY1ZmYzNmM2ZDQ5NzgzZDJhMDYzMjVlNGFlM2U4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qNv256zVNsOjG0fKjFR43BEnjXpYuOfp5eDw4SDJG84)

你有没有对他的代码进行完善？

xixi88888888, 2024-09-10:
> 非常感谢。我根据项目： (https://github.com/yangfan852219770/beidou-sdr-sim) 和GPS-SDR-SIM调制出了B1I信号，但是我在使用动态坐标生成的信号会出现导航电文读错的情况，但静态坐标不会，而且计算的伪距错误。 ![屏幕截图 2024-08-28 [...]
你有没有对他的代码进行完善？还有我也想问下你这个是什么软件接收机？

cehuidong, 2024-09-10:
> 可以一样调制，仅仅B1I可看做bpsk
>
>
> 非常感谢。我根据项目： (https://github.com/yangfan852219770/beidou-sdr-sim) 和GPS-SDR-SIM调制出了B1I信号，但是我在使用动态坐标生成的信号会出现导航电文读错的情况，但静态坐标不会，而且计算的伪距错误。 ![屏幕截图 2024-08-28 [...]

我对照GPS-SDR-SIM对D1导航电文进行了完善，以及更新子帧那里。我使用的是GNSS-SDR软件接收机和GNSS-SDR-monitor监视器。

cehuidong, 2024-09-10:
> Interesting constellation and Doppler diagrams. Which software receiver did you use to verify the simulated BeiDou signals?

> software receiver

Thank you for the help with your code! The software I'm using is a GNSS-SDR software receiver and a GNSS-SDR-monitor. The simulated GPS signals are perfectly solved, but the BeiDou signals show the above problem.

xixi88888888, 2024-09-10:
> 我对照GPS-SDR-SIM对D1导航电文进行了完善，以及更新子帧那里。我使用的是GNSS-SDR软件接收机和GNSS-SDR-monitor监视器。

你可以用离线的软件接收机去解析试一下，看看是不是子帧的更新有问题

cehuidong, 2024-09-11:
> 你可以用离线的软件接收机去解析试一下，看看是不是子帧的更新有问题

我准备寻找其他软件接收机测试一下。GNSS-SDR对北斗B1I信号进行测试时可能存在一个70-80 ms的信号传播时延，导致有一个相当大的接收机钟差，会让伪距存在负值，但不影响定位。
可能是我调制的信号能量较低或初始化相位的问题，导致无法定位。