osqzss/gps-sdr-sim #408: Geolocation does not change

Keyt13, 2024-08-15:
I use hackrf one great Scott gadgets (ANT500 / GPS ACTIVE ) antennas. (R8)
Checking through Android applications, gps signals are coming. Geolocation does not change. 
I did it on Kali Linux , android , windows .

The program always disconnects itself when total time 300.xxx
[image]
[image]

Keyt13, 2024-08-15:
[image]
[image]

victor-vu-max, 2024-08-19:
Same here, really stressful

victor-vu-max, 2024-08-19:
[image]

victor-vu-max, 2024-08-19:
[image]

Robotix-00, 2024-08-19:
This is intended behavior, as gps-sdr-sim only generates a 300 second spoofed signal by default.
If you need more time, you can use the **-d <sec>** option when generating the .bin file and specify up to 86400s (24h).

Note that this only applies to static mode.

victor-vu-max, 2024-08-19:
[image]
[image]
[image]
[image]
Geolocation does not change even i switched to flight mode on mobile

Robotix-00, 2024-08-20:
You are not using external attenuator. The readme describes the hardware like so:
"The TX port of a particular SDR platform is connected to the GPS receiver under test through a DC block and a fixed 50-60dB attenuator."
Those are parts that need to be put in between the sdr and the antenna to achieve the right result. GPS is a very weak signal, as the satelites are quite far away, so your sdr is way stronger than your receiver could ever reasonably expect.

victor-vu-max, 2024-08-20:
I have read about how to degrade the GPS signal emitted from the R8 antenna from the SDR. However, the GPS data on current Androin or IOS phones uses A-GPS technology and it depends on Wi-Fi or Bluetooth connections. Therefore, GPS spoofing is completely impossible with these mobile devices.

Keyt13, 2024-08-21:
Thank you all.
With the help of Android apps I managed to change it, but the coordinates I specified are as different as possible from what I was given. 

Can you tell me what might be affecting this?

Keyt13, 2024-08-21:
[image]
[image]
[image]
[image]
[image]

Keyt13, 2024-08-21:
Thank you all for responding. I did it!

closed 2024-08-21 by Keyt13

Robotix-00, 2024-08-22:
You have spaces in between your location when generating the signal. Gps sim interprets those as different attributes. If you remove the spaces everything should work.