osqzss/gps-sdr-sim #412: Error opening PLUTOSDR TX device: Success (0)

Code-Sierra, 2024-09-09:
[image]

osqzss, 2024-09-13:
Sorry, I am not a Linux person and couldn't replicate the issue on my Windows machine.

Code-Sierra, 2024-09-13:
I would have got the same issue there also right. I got my gpssim.bin file, but the plutoplayer is not transmitting. It says the above mentioned error. 

Is it working fine with the windows machine?

osqzss, 2024-09-13:
It just works fine. You can ignore libxml 2 errors.
[image]

osqzss, 2024-09-15:
I got the same error on a Ubuntu machine. It seems a libiio issue, and I can't solve the problem.

osqzss, 2024-09-15:
Remove iio_create_default_context (line 128) from plutoplayer.c and rebuilt it. This should solve the problem.

Code-Sierra, 2024-09-15:
It's running after removing line 128. But, There is no change in location.

Code-Sierra, 2024-09-16:
> Remove iio_create_default_context (line 128) from plutoplayer.c and rebuilt it. This should solve the problem.

What's the purpose of this line of code? If it works in windows, why should we remove the line of code  for linux? Can you show a example of the location changing.

Code-Sierra, 2024-09-17:
It works after removing line 128 in linux.

Code-Sierra, 2024-09-17:
How to build plutoplayer for windows? Which one should I use - plutoplayer_win or the one which is available in player?