osqzss/gps-sdr-sim #413: Recent BRCD files not (always) working

apppie123, 2024-09-12:
Strangest issue:
I've been using your code for a year now and it worked perfectly.
However, since recently it is giving me random issues.

Attached two brdc files downloaded from nasa.
brdc1150.24n.gz works as expected.
brdc2550.24n.gz sometimes works, sometimes I get an error at the start the the end time is incorrect. Max end time is some random numbers and not a date. However, when I restart then sometimes it does work.

As said, this compiled binary is running for a year without changes. Running on Ubuntu 22.04.

I also tried the multi-sdr-sim code and it has the same problem. brdc2550 gived me a segmentation 
fault, brdc1150 works.

I cannot find any obvious differences in the file.

[brdc2550.24n.gz](https://github.com/user-attachments/files/16978057/brdc2550.24n.gz)
[brdc1150.24n.gz](https://github.com/user-attachments/files/16978058/brdc1150.24n.gz)

osqzss, 2024-09-12:
It could be the same issue: https://github.com/osqzss/gps-sdr-sim/issues/400

apppie123, 2024-09-12:
> It could be the same issue: #400

Yes! Thank you so much for your super fast reply.
I set it from 13 to 15 and now it runs again with 2550.

I see you made an update 3 months ago fixing this, I totally missed it.

Thank you so much for the hard work!

Code-Sierra, 2024-09-13:
Error opening PLUTOSDR TX device: Success (0)
Any solution for this? #412

closed 2024-09-15 by osqzss