osqzss/gps-sdr-sim #414: I'd like to add support for GPS L2 signals - need some direction.

bangpradyumna, 2024-09-13:
Hello @osqzss , 
I've recently been introduced to the wonderful world of GNSS and the gps-sdr-sim works perfectly for simulating GPS L1 signals. 

I would really love to contribute to this repo and learn more about how GPS signals are formed, and for that I feel a good way is to implement GPS L2 functionality in this repository.

I must admit that I'm a bit overwhelmed by the code but I was wondering if I could get some directions on how to go about implementing GPS L2.

Based on my research, this is what i feel is required for GPS L2 support in current code :

1. Carrier Frequency: Add support for 1227.60 MHz for L2.
2. PRN Codes: Implement L2 CM/CL PRN code generation.
3. Modulation: Modify the modulation logic for L2 (BPSK with CM/CL codes and CNAV data).
4. Configuration: Generate dual frequency IQ files and test them with the SDR's and receivers. _(Question - Would I need to use dual channel SDR for L1 and L2 generation or the same channel will be able to transmit in both frequencies ?)_

Please let me know if I've missed anything in the above points, if you have any recommendations - I'd really appreciate them.

Thanks!

osqzss, 2024-09-16:
Thank you for your contribution. I have just made a 'develop' branch for these new signals. Please push your commits to it. Regarding your question 4, there is no dual-channel SDR available a reasonable price. You need separate IQ files and a pair of SDR devices for L1 and L2 signals.

bangpradyumna, 2024-09-19:
Thanks @osqzss, I'm using a USRP B210 which is dual channel so I can test it.

However, What do you think about the approach that I've described ? Is that a good one to start with or am I missing something there ?

slblbwl, 2024-11-12:
Hello, I also want to use GPS-SDR-SIM to transmit L2 GPS signals. I would like to discuss with you how to transmit, is that okay? Translate Hello, I also want to use GPS-SDR-SIM to transmit L2 GPS signals. I would like to discuss with you how to transmit it, is that okay?