osqzss/gps-sdr-sim #415: How to build plutoplayer for windows?

Code-Sierra, 2024-09-17:


osqzss, 2024-11-09:
Added readme file: https://github.com/osqzss/gps-sdr-sim/tree/master/player/plutoplayer_win

Code-Sierra, 2024-11-13:
I got errors while doing with Visual studio 2022. I tried using **gcc** .

`C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x2e5): undefined reference to `_imp__iio_create_default_context'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x304): undefined reference to `_imp__iio_create_scan_context'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x31f): undefined reference to `_imp__iio_scan_context_get_info_list'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x336): undefined reference to `_imp__iio_context_info_get_description'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x353): undefined reference to `_imp__iio_context_info_list_free'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x360): undefined reference to `_imp__iio_scan_context_destroy'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x375): undefined reference to `_imp__iio_context_get_devices_count'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x3aa): undefined reference to `_imp__iio_context_find_device'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x3c3): undefined reference to `_imp__iio_device_find_channel'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x3cc): undefined reference to `_imp__iio_channel_attr_write_bool'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x42c): undefined reference to `_imp__iio_channel_disable'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x43d): undefined reference to `_imp__iio_channel_disable'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x446): undefined reference to `_imp__iio_context_destroy'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x4d6): undefined reference to `_imp__iio_context_find_device'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x508): undefined reference to `_imp__iio_device_set_kernel_buffers_count'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x525): undefined reference to `_imp__iio_device_find_channel'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x556): undefined reference to `_imp__iio_channel_attr_write'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x573): undefined reference to `_imp__iio_channel_attr_write_longlong'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x5bb): undefined reference to `_imp__iio_channel_attr_write_double'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x5e2): undefined reference to `_imp__iio_channel_attr_write_bool'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x6a5): undefined reference to `_imp__iio_channel_enable'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x6d5): undefined reference to `_imp__ad9361_set_bb_rate'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x6fe): undefined reference to `_imp__iio_device_create_buffer'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x754): undefined reference to `_imp__iio_buffer_start'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x795): undefined reference to `_imp__iio_buffer_push'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x809): undefined reference to `_imp__iio_buffer_destroy'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x834): undefined reference to `_imp__iio_create_network_context'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x85a): undefined reference to `_imp__iio_strerror'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x8da): undefined reference to `_imp__iio_strerror'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x8f7): undefined reference to `_imp__iio_device_find_channel'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x910): undefined reference to `_imp__iio_channel_attr_write_bool'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x9a3): undefined reference to `_imp__iio_create_context_from_uri'
C:\Users\VTPL-S~1\AppData\Local\Temp\ccQPP2wl.o:plutoplayer.c:(.text.startup+0x9b7): undefined reference to `_imp__iio_create_network_context'
collect2.exe: error: ld returned 1 exit status`

How to solve this error? 

This is what I ran in the plutoplayer_win directory.

`gcc -lm -O3 -mconsole -o plutoplayer plutoplayer.c`

Code-Sierra, 2024-11-14:
I have built using Visual Studio 2022. I had missed some steps.

Code-Sierra, 2024-11-14:
But, if you could put a readme for the instructions on how to build using gcc compiler, it would be helpful.

liuyongcong, 2025-01-08:
> 但是，如果你能放一个自述文件来说明如何使用 gcc 编译器进行构建，那将会很有帮助。

I am using VS2022 to build the PLUTO-WIN program, but it failed. Can you provide me with your build output?

liuyongcong, 2025-01-08:
> I have built using Visual Studio 2022 successfully. I had missed some steps.

I am using VS2022 to build the PLUTO-WIN program, but it failed. Can you provide me with your build output?