osqzss/gps-sdr-sim #419: Unable to detect transmitted signals

Octoshroom, 2024-10-25:
Hello!

I am currently in the process of trying to get this code working, but despite following the instructions I have had no luck.

My setup is as follows: 
**Transmitter**
HackRF One, firmware version 2018.01.1, hackrf version 2023.01.1
ANT500 antenna (can transmit signals from 75 MHz - 1 GHz, variable length)

**Receiver**
Colorado 300 GPS

The Colorado 300 has no problems receiving real GPS signals when brought outside, even if it had detected none when started inside, but is completely unable to detect the signals transmitted by the HackRF. I do not yet have access to a 20dB attenuator, something that I have seen be recommended in other posts,  but have tried other methods of reducing signal strength such as having multiple walls between the transmitter and receiver, changing antenna length, wrapping the antenna in aluminum foil and so on. I do not currently have access to any ways of detecting or validating the signals other than the GPS as I am just getting started playing around with SDRs.

I am using the latest ephimeris file, updated whenever a new one is released, and have tried transmitting using several adjacent frequencies in case of frequency drift.

The command to generate the gps-signal-file is 

`./gps-sdr-sim -b 8 -e [location of ephimeris file] -l 55.7558,37.6176,156 -0 gps.bin`

The command I use to transmit is 

`hackrf_transfer -t gps.bin -f 157542000 -s 2600000 -a 1 -x 0`

Any help would be appreciated, because as of now nothing I have tried has provided any results.

Octoshroom, 2024-10-28:
Having done some further testing with a spectral analyzer, it is clear that the device is transmitting, and that it is transmitting at the correct frequency. However, even when the HackRF TX port was connected to a GPS receiver directly with a wire, the receiver was unable to receive a signal. Any advice as to what may be going wrong with the gps.bin generation (as I assume that is where the problem may well be) would be appreciated.

osqzss, 2024-10-29:
It seems something wrong with the 8-bit option. I'll look into it.

osqzss, 2024-10-29:
The center frequency of your hackrf_transfer command was wrong. It should be 1575420000.

Octoshroom, 2024-10-31:
Am looking to find an external clock to see if that has any effect on the success. How precise does it have to be, and do you have any recommendations?

Have tried adding a 20 dB attenuator without any luck, and have managed to confirm that the device itself is transmitting something using a RTL-SDR. However, even when connected to it directly by wire the receiver is unable to pick up the signals. 

Have used 1575420000 Hz as the center frequency, saw after your response that I had written the wrong frequency in the original post. My apologies. I have double checked the command I use has the correct center frequency, so that should not be the cause. I am guessing that it is the external clock that is needed, as I see no other reason as to why it would not work

Octoshroom, 2024-10-31:
Thank you! I will try to get my hands on one and report back once I've tried it

Octoshroom, 2024-11-07:
Adding a more precise external clock solved the issue. Thank you for the advice

closed 2024-11-07 by Octoshroom