osqzss/gps-sdr-sim #420: But  isn't it enough if the 5th subframe is sent only the first time  of the total duration? If we do it everytime while we send the navigation message is sent - aren't we sending like 5-1-2-3-4-5,5-1-2-3-4-5,5-1-2-3-4-5...etc .Wouldn't this be wrong ? Great work with the code by the way .

virat-fan18, 2024-10-27:
But  isn't it enough if the 5th subframe is sent only the first time  of the total duration? If we do it everytime while we send the navigation message is sent - aren't we sending like 5-1-2-3-4-5,5-1-2-3-4-5,5-1-2-3-4-5...etc .Wouldn't this be wrong ? Great work with the code by the way .

_Originally posted by @virat-fan18 in https://github.com/osqzss/gps-sdr-sim/issues/180#issuecomment-2440135438_

cehuidong, 2024-11-01:
> dwrd[60] - But isn't it enough if the 5th subframe is sent only the first time of the total duration? If we do it everytime while we send the navigation message is sent - aren't we sending the subframes like 5-1-2-3-4-5,5-1-2-3-4-5,5-1-2-3-4-5...etc .Wouldn't this be wrong as It should be 5-1-2-3-4-5-1-2-3-4-5-1-2....etc(as you meantioned in aprevious issue thatwe have to simulate for a time before the transmission time ) ? Great work with the code by the way .
> 
> _Originally posted by @virat-fan18 in [#180 (comment)](https://github.com/osqzss/gps-sdr-sim/issues/180#issuecomment-2440135438)_

Because dwrd 9 and 59 are identical. I don't know if I'm right, but I think the principle of the part is as shown in the picture：
[image]

xixi88888888, 2024-12-02:
> dwrd[60] - 但是如果在整个持续时间内只发送第 5 个子帧，这还不够吗？如果我们每次发送导航信息时都这样做，难道我们不是在发送 5-1-2-3-4-5、5-1-2-3-4-5、5-1-2-3-4-5...等子帧吗？这难道不对吗？它应该是 5-1-2-3-4-5-1-2-3-4-5-1-2...等（正如您在上一期中提到的，我们必须在传输时间之前模拟一段时间）？顺便说一句，代码写得很好。
> 
> _最初发布者@virat-fan18在[#180 （评论）](https://github.com/osqzss/gps-sdr-sim/issues/180#issuecomment-2440135438)_

Because this code updates the navigation messages of all satellites in all channels simultaneously, the navigation messages of all satellites are not aligned when received. If all updates are made according to the fastest updating channel, it may cause some slow channels to transmit the next frame's position message. Therefore, using 60 to save an additional subframe (6s) can completely cover the updates of all message frames

cehuidong, 2024-12-02:
> Because this code updates the navigation messages of all satellites in all channels simultaneously, the navigation messages of all satellites are not aligned when received. If all updates are made [...]

我看到代码中是将缓冲放在了最前面，并不是让第一子帧的时间为开始时间，这在GPS中没影响吗？还是我理解错误。

xixi88888888, 2024-12-05:
没有呀，代码做了机制，电文更新时候导航电文的索引也会变化，比如之前传输的索引是57，电文更新之后就变为7了

cehuidong, 2024-12-05:
> 没有呀，代码做了机制，电文更新时候导航电文的索引也会变化，比如之前传输的索引是57，电文更新之后就变为7了
> […](#)

我可能没表述清楚，我指的是子帧时间更新那部分。我之前疑惑为什么第一个子帧传递的不是起始时间，GPS的子帧传递的周内秒应该是下一个子帧的起始时间对吧。

xixi88888888, 2024-12-06:
你可以仔细看下tow的定义，周六午夜整点更新的时候它不是从0开始的，而是从1开始的。

cehuidong, 2024-12-06:
好的，谢谢，我明白了。

 
 
 
 
 
 
 
    
    
   
&mdash;

xixi88888888, 2024-12-12:
哥们，北斗的你有没有模拟geo卫星的，我这边geo卫星的模拟感觉按照gps sdr sim的方法，有很大出入呀

 
 
 
 
 
 
   
   
   
   
   
   
   
      
      
     
 &amp;mdash;

cehuidong, 2024-12-12:
> 哥们，北斗的你有没有模拟geo卫星的，我这边geo卫星的模拟感觉按照gps sdr sim的方法，有很大出入呀
> […](#)
> ---Original--- From: ***@***.***&gt; Date: Fri, Dec 6, 2024 21:53 PM To: ***@***.***&gt;; Cc: ***@***.******@***.***&gt;; Subject: Re: [...]

我只模拟了MEO/IGSO卫星。你指的出入很大是指哪方面的，最终定位结果吗？

xixi88888888, 2024-12-12:
是的，加入了geo进行解算之后，解算的速度错误，定位结果变差

cehuidong, 2024-12-12:
> 是的，加入了geo进行解算之后，解算的速度错误，定位结果变差
> […](#)

你用什么硬件测试的。你可以用GNSS-SDR验证一下，看解调出的导航电文内容是否正确。或者是传递的时间有点问题。

xixi88888888, 2024-12-12:
我用b210测试的，我先用gnsssdr测试一下，到时候再交流