osqzss/gps-sdr-sim #421: There are some dynamic limitation to create simulations?

Derso1, 2024-11-07:
Hi!
I whould like to know if there is some dynamic limitation in the gps-sdr-sim to create cenarios with hight velocity

I am trying to simulate a rocket, but my rocket will have 7000 m/s of velocity (constant velocity).

The code worked. I got the *.bin file, but I dont know if the bin file is correct.

Someone know to help in this issue.

Tks.

Derso1, 2024-11-07:
Tks. @osqzss!!!!
In the begin I created my simulation using the rocket.csv that is native in the gps-sdr-sim.

When I used the rocket.csv the *.bin file that I got everythig worked fine. 

After this I created a new simulation. This simulation has a big aceleration in the beginning, but, in the middle of the flight there is only hight velocity (but constant)

My question is:

The *.bin file for the middle of the flight will work?

Tks

osqzss, 2024-11-07:
There should be no problem.