osqzss/gps-sdr-sim #424: Checking for Multipath Usage in Simulation

jpereiramendoza, 2024-11-24:
Hi!!

I have a question about multipath in the simulation generator. Is it included in the simulator file? Can I control the amount of multipath in my simulation? If I want to verify this in the code, which section should I check?

I checked the list of issues, but I couldn’t find any information about it.

Thanks in advance! 

Regards

Sselion, 2024-12-02:
Hi.

With my knowdlege of this code, there is no simulation of multipath. I would like to point out that it has been some time since I studied this simulation.
Basically how this signal simulation works: 

1. Read Rinex
2. Base on rinex compute position of satellites and range/pseudorange between satellite and receiver.
3. Compute gain of signal for each satellite base on pseudorange
4. Base on pseudorange compute freq. delay and compute 100 ms of IQ samples one by one.
5. repeat from 2.

Somewhere is check of sat visibility and gps message computation, but I don't thing it's important for this question.

I hope my answer helped you somehow.
With regards

jpereiramendoza, 2024-12-02:
Hi !!

Thanks for your answer.

Currently, I’m working on mathematical models to reduce multipath effects on GPS signals. Therefore, I need some files with these effects to check whether my algorithm works or not :D. For this reason, I think it’s a good idea to include a multipath option in the simulator. The first step is to check whether this problem is included or not.

For your comment, This simulator doesn´t has the multipath problem. I'll check if I can implements it. 

Thanks for your time and your answer ! 

Regards
Jota

cdhmanning, 2025-01-14:
The code does not support multipath, but I would think it is fairly easy to add it.

What sort of muti-path effects are you trying to simulate?

jpereiramendoza, 2025-01-14:
> The code does not support multipath, but I would think it is fairly easy to add it.
> 
> What sort of muti-path effects are you trying to simulate?

Hi !!

Thanks for your answer. 

In this moment, I don't have information about what multi-path effect  I can trying to simulate. First step is check if the simulator has the implementation about it.

 I'll check in the future it, because now i don't have time, but I want to implement it. I think that is a very nice  

Regards !!

closed 2025-01-14 by jpereiramendoza