osqzss/gps-sdr-sim #425: GPS Receiver Inconsistent Position Lock Near Adalm Pluto SDR

Code-Sierra, 2024-12-11:
I am using the Adalm Pluto SDR for transmission and have observed that the GPS receiver, when placed near the Adalm Pluto TX antenna, exhibits inconsistent behavior. At times, it locks the position, but other times it fails to do so. This issue occurs specifically when the TX antenna is in close proximity to the GPS receiver.

I have ensured that no external GPS signals are present within the test area. Additionally, I am not using any attenuators or clock references in the setup.

I suspect the issue might be related to clock instability, but **I am looking for clear reasons and practical improvements to ensure the GPS receiver achieves a consistent lock every time I run this program.**

Code-Sierra, 2024-12-11:
Please suggest the external clock reference module's Model number and brand or even the purchase link.

bangpradyumna, 2024-12-17:
https://www.mictronics.de/posts/ADLAM-Pluto-OCXO-Mod/

Do the second mod suggested here

Code-Sierra, 2024-12-18:
I got the leo bodnar gpsdo device. Is it fine?

osqzss, 2024-12-18:
Yes, they have good GPSDOs.

Code-Sierra, 2024-12-19:
Any way to bring in Beidou as well? or Any method to increase the number of satellites?

cdhmanning, 2025-01-14:
If it only fails when the antenna is too close to the GPS receiver it could be that the transmit power is too high causing the GPS front-end to saturate which would cause distortion.

Try to either turn down the TX power by a few dB or fit an attenuator.